Pitt-RAS/micromouse-2016 #5: Create Sensor Class

QuentinTorg, 2015-12-15:
Create a new sensor class in sensors_range.cpp that will be instantiated once per sensor.  The input for the instantiation will at least need a pin number as an input, which will come from conf.h.  The function for the class won't have any input parameters, but will output a floating point range from the sensors in millimeters.   

The sharp rangefinders have an analog output that varies with distance.  The output doesn't vary linearly, so either research or testing should be done to model the output vs. range, and find a best fit equation that can be used to get a reasonably accurate output.  

Our Teensy only takes analog inputs from 0-3.3 volts, but the Sharp rangefinders output from 0-5 volts.  A voltage divider is already in our board to reduce the voltage output from the rangefinder.

QuentinTorg, 2015-12-15:
The distances measured should be from the center of the robot instead of from the actual rangefinder.  This can be done with a simple offset that would be part of the input parameters during the instantiation of the class.  The offset values should be stored in conf.h

amiller27, 2015-12-15:
Does this really need to be a class? It isn't storing any state between function calls, so it seems like it might be simpler to just write a function that you would call with the pin number, e.g. get_range(RANGE_FRONT) with the pins defined in conf.h

QuentinTorg, 2015-12-15:
I think we're going to have to input a pin number and an offset so that we can always get data from the center of the mouse.  Standardizing it to the center of the mouse means changing sensor configurations won't affect performance very much if we ever switch them up.

I don't know if that warrants the use of classes in itself, but I saw it more as a thing of style consistency.  right now we have most of our functions declared but not yet defined, so its easy to make them classes if they need to be.  The code that is currently used is already uses classes heavily, so I wasn't opposed to just going all classes

QuentinTorg, 2015-12-15:
Also, if there's ever a chance that we might need recalculations during instantiation I think classes are the way to go.  I don't know if we will need any of that in the rangefinder classes/functions though

amiller27, 2015-12-15:
The offset can be determined based on what pin you're asking for, e.g. RANGE_FRONT gets RANGE_FRONT_OFFSET, etc.  Were there any recalculations you had in mind?

QuentinTorg, 2015-12-15:
I like that plan for it pulling the offset after the fact.  I didn't have any calculations in mind yet

closed 2016-01-31 by xaviertorgerson