Pitt-RAS/micromouse-2016 #7: Standardize variable and function naming conventions

QuentinTorg, 2015-12-15:
Our variables and functions all have different naming conventions.  Let's decide on a correct format and post it somewhere for reference.

Also, the way classes are instantiated and used should be the same.  Example, The motionCalc class and the PIDcorrectionCalculator classes both work the same way, but are instantiated and used differently.  If one way is better than the other let's do it that way.  If they're identical, lets decide which way the group like's best and do it that way

amiller27, 2015-12-28:
I vote that we follow the [Google style guide](https://google.github.io/styleguide/cppguide.html#Naming)'s naming conventions, **except** that class methods should be lower camelcase, not upper camelcase.  Thoughts?

xaviertorgerson, 2015-12-28:
I vote what Aaron said.

ryanpitt, 2015-12-29:
Works for me.

We also ought to settle on one kind of indenting. Right now there's an unsightly mix of two space indents, four space indents, and real tabs.

QuentinTorg, 2015-12-29:
My vote for the indent is that we go in the arduino IDE and right click and hit auto format. It standardizes for us. I think different editors just have different number of spaces for the tab, so everyone can just keep using their tab whatever size it is, and whenever it gets down to a final commit just bring it over to arduino and do the autoformat and it will standardize.

ryanpitt, 2015-12-31:
Well whatever we do, I just verified the Arduino IDE uses two spaces for indents (and not a tab displayed as two characters wide).

amiller27, 2015-12-31:
Let's do two space indents then.

closed 2016-03-09 by amiller27