Pitt-RAS/micromouse-2016 #8: Add accelerometer functionality

QuentinTorg, 2015-12-15:


QuentinTorg, 2015-12-15:
The magnetometer may be too greatly affected by our motors and encoders to give us usable data

QuentinTorg, 2015-12-28:
During a straight forward motion, the drift of the gyro could be removed by taking into account what the centripetal acceleration should have been during that motion.  If the gyro says that there was a rotation, but the accelerometer saw no centripetal acceleration during that time, then the rotation that the gyro saw should be removed or at least made smaller than was measured.

amiller27, 2016-01-04:
Gyro drift probably isn't going to be a problem, and if it is, drift in the gyro is going to be comparable to the noise level in the accelerometer.  The gyro is currently drifting by about a hundredth of a degree every few seconds while stationary, which would require accelerometer noise less than on the order of a micro g.  We could put a low pass filter on the accelerometer to get the noise level down a little, but I think the gyro alone should be enough.

xaviertorgerson, 2016-01-04:
Probably won't get it better than a hundredth of a degree drift unless you get the earth to stop spinning.

QuentinTorg, 2016-03-17:
Can we create a gyro_rotate(angle) function that uses only the gyro to make pivot turns of the inputted angle?

we can decide on a reasonable rotational velocity for a turn and that will be a set parameter that doesn't change.  Then it will work similarly to motion_forward in that it accelerates up to the rotational velocity, moves, then decelerates to 0, stopping at the final angle.

It should try to match the implementation that will be used in motion_forward as closely as possible as far as forwarding gyro error into a PID function, that generates an encoder offset.

closed 2016-03-27 by amiller27