Pitt-RAS/micromouse-2016 #20: Move initialization into a function

ryanpitt, 2016-02-07:
In response to us forgetting to set the analog write resolution in the testing branch, we decided it makes sense to be able to do all initialization with one function call so that it's clear what needs to be initialized to make the rest of our code work properly.

Things that should be in there include at least setting the analog frequency, setting the analog resolution, and zeroing the encoders.

QuentinTorg, 2016-02-08:
Should this be part of the conf file?

amiller27, 2016-02-08:
I don't think there should be any code in conf.  We could just mark it with strongly worded comments in the sketch to separate it from the stuff that's temporary

ryanpitt, 2016-02-09:
Yeah, actually micromouse.ino is gettting pretty small already in the master branch. We probably don't need to worry about this issue for testing stuff.

closed 2016-02-09 by ryanpitt