Pitt-RAS/micromouse-2016 #27: Improve isWall()

xaviertorgerson, 2016-03-09:
The current isWall() uses a Schmidt Trigger to determine if there is a wall there that can be followed.  This system is very good for driving straight between two walls.  The problem is when one wall falls out sometimes it will take a second to trigger the Schmidt Trigger which means it will continue to follow the wall that fell out for a few moments.  After tending towards the wall that just fell out for a few seconds the trigger realizes there is no wall there, but because it has moved away from the actual wall the Schmidt Trigger is often set off on the opposite side.  The wall and the dropped away wall both end up crossing the Schmidt threshold at about the same time so it's 50/50 which one it follows.

ryanpitt, 2016-03-09:
Partial solution is to ignore range sensor error while we're near the center of the cell, which prevents us from taking readings at wall edges. I tried implementing that, and it _did_ help after some adjusting. But it seems like the generated error for one wall isn't working properly, even for a one-wall-only test.

QuentinTorg, 2016-03-13:
Currently we are ignoring readings around the vertices, and only correcting during the middle 50% of each cell.  Functionality should be added to the driver that says if the robot knows there is a wall in the current cell and the next cell, it can continue to correct with the rangefinder even though it is crossing over a vertex.

We also need a second isWall() function.  Currently the motion code uses the same isWall thresholds that the navigator uses for mapping.  The navigator needs to be more robust than the motion code though, and should probably have a larger threshold than the motion code.

QuentinTorg, 2016-03-13:
This may be all that we need to get rid of our jitter while transitioning from wall to no wall.  
http://micromouseusa.com/?p=417

He has a middle value, and uses the sensor for centering if it is returning a range that is less than the middle value.  This means that as our walls drop away they will no longer be looked at, and they won't be looked at again until they are within our center value.  Another repercussion is that most of the time our rangefinders will probably only follow one wall.  It probably won't be an issue, but its something to keep in mind

QuentinTorg, 2016-03-17:
Green Ye's wall following logic.  http://micromouseusa.com/?p=389

QuentinTorg, 2016-04-03:
Green Ye is the man

closed 2016-04-03 by QuentinTorg