Pitt-RAS/micromouse-2016 #33: Drive Diagonals

QuentinTorg, 2016-03-13:
Major speed improvements will come from driving diagonals during chaos mode.  We need to do some research to find out how other people do side to side error correction while driving diagonals.  On shorter diagonal paths we could probably get away without any rangefinder error correction.  Using encoders only our turns are very accurate, so we will only improve after adding gyro, and better calibration to make sure we always start in the center of the cell.  

The driver will need to be updated to incorporate diagonals into a path.

QuentinTorg, 2016-03-13:
Here's some starting information: http://www.micromouseonline.com/2009/04/23/ternary-numbers-for-diagonal-micromouse-turns/

http://www.micromouseonline.com/2014/05/05/diagonal-paths-micromouse-using-state-machine/

QuentinTorg, 2016-03-14:
http://www.sensorsportal.com/HTML/DIGEST/may_2014/Vol_170/P_2035.pdf
This paper discusses using just the forward sensors to drive diagonals.  If the robot is pointed towards a peg while driving a diagonal then it will read a closer range than is acceptable.  The robot can use this similarly to driving between parallel walls in order to avoid upcoming pegs.  

We would have peaks and valleys to our readings as pegs go past, so we would only correct if our range values were so close that we knew we would be hitting a peg.  

We may also be able to use our derivative functions to remember our peak readings out of each side sensor, and use those to make an educated guess at how much correction we would need to remain centered

xaviertorgerson, 2016-03-26:
Diagonals are only required in a chaos mode really.  We can have a state where the sensors don't really center us but they just keep us from hitting walls if we are too close.  The rest we just count on our motion function.  The sensors can just keep us from catching an edge.  This is the new current logic for the sensors pretty much as far as I'm aware its just that in straight paths the buffer for the center is much tighter we can just give it a larger buffer.  Thoughts?

closed 2016-04-11 by ryanpitt