Pitt-RAS/micromouse-2016 #35: Update Arduino Installations

QuentinTorg, 2016-03-17:
The robot should always have the same program and settings no matter which computer we upload from. This means everyone should have the same arduino IDE, teensyduino version, and overclocking settings enabled.

Use:
-Arduino 1.6.7
-Teensyduino 1.27
-Update all libraries Including EncoderMod and accelerometer libraries 
-Follow overclocking instructions to enable '144MHz optimized'. Do not enable higher speeds than 144. The arduino IDE will default to your highest enabled speed, and 144MHz is considered stable
http://arduino.stackexchange.com/questions/3908/teensy-3-1-no-overclock-option-for-168-mhz

Comment below when you update

amiller27, 2016-03-20:
Done

ryanpitt, 2016-04-06:
Done

ryanpitt, 2017-01-13:
Since people are in the process of setting up build systems again. How about...

 - Arduino 1.8.0
 - Teensyduino 1.34
 - Teensy 3.1/3.2 144 MHz

We would be well served to have a README of sorts with this information, including how to overclock to 144 MHz.

ryanpitt, 2017-01-31:
We're enforcing patched library versions and clock speed. README overclocking instructions in #101. Not enfocing Arduino/Teensyduino versions.

closed 2017-01-31 by ryanpitt