Pitt-RAS/micromouse-2016 #65: Path Parser ends a square early

amiller27, 2016-08-28:
[image]

When the robot runs through the above maze, starting in the bottom left and going to the top right, the path parser gives a solution which makes the robot stop in the cell one south of the target

awglyde, 2017-01-28:
I found some use cases that fail to parse the motion correctly.

The original issue use case:
kNorth, kEast, kNorth, kNorth

A use case I found while debugging (only gets to the first east, ignores everything else):
kNorth, kNorth, kEast, kNorth, kEast, kEast, kEast

ryanpitt, 2017-01-30:
If I remember correctly, we thought this bug only occurred when the robot enters the finish cell from a straightaway. I wonder if what's actually happening is the path parser is just stopping at the last turn. That would be consistent with our previous guess and with your new observations.

awglyde, 2017-03-02:
The original assumption is close to accurate. When the mouse exits a turn and needs to move forward it fails to do the last move because of the underlying queue structure. It pops a North (i.e. forward) off the queue to determine how to exit but never pushes a forward.

You saw this as missing the last move in the maze photo above because there were a series of forwards. It was truly missing the first forward of the series not the last. Thus the logic for the diagonals was the problem.

I've attach photos of the some problem mazes (warning: I drew them on a white board):

[20170302_142639](https://cloud.githubusercontent.com/assets/7102050/23524864/7170f716-ff5a-11e6-9ad5-6168547d7b2d.jpg)
[20170302_143503](https://cloud.githubusercontent.com/assets/7102050/23524863/716f24e0-ff5a-11e6-829b-ad3d8d89586f.jpg)
[20170302_144804](https://cloud.githubusercontent.com/assets/7102050/23524862/716f0e56-ff5a-11e6-9bc8-43c55598cac1.jpg)

It would be a good idea to test those explicitly as well as some random maze configurations. Likely those that have diagonals as that is what my fix affects.

ryanpitt, 2017-03-04:
Three test cases that don't work on 39528ea926f2fff2b0cda3ac0a8e036aa2174db0.

[image]

[image]

[image]

ryanpitt, 2017-03-14:
See #122.

closed 2017-03-14 by ryanpitt