Pitt-RAS/micromouse-2016 #78: I2Cdev library is patched under the same name as upstream

ryanpitt, 2017-01-13:
Users who have upstream I2Cdev installed can't build the project.

Make a fork "I2CdevMod" and modify our code to use that instead of upstream.

ryanpitt, 2017-01-13:
Also need to confirm that this is the build issue we were having, and that Arduino 1.8.0 and Teensyduino 1.34 are _not_ the problem.

asaba96, 2017-01-13:
I can confirm that it builds just fine with Arduino 1.6.12 and teensy 1.34, as well as with Arduino 1.8.0, both times with the modified library. I believe this problem was due to the library, not the newer versions.

ryanpitt, 2017-01-13:
Confirmed working with Arduino 1.8.0 and Teensyduino 1.34 on Linux.

ryanpitt, 2017-01-13:
Whoops. In turn, the MPU9150 lib uses I2Cdev. So I'm going to have to patch that to refer to I2CdevMod.

Unless somebody has a better idea.

amiller27, 2017-01-13:
Ideally we would fix the actual problem in I2Cdevlib which prevents it from dealing with multiple i2c buses so that our solution would be backwards compatible with I2Cdevlib, but that would be hard.

ryanpitt, 2017-01-13:
Agreed.

ryanpitt, 2017-01-16:
@amiller27: I used your idea for renaming via macros. I couldn't find a way to make a conventional diff patch that renames files with changes like git does. But I _was_ able to make concise git commits on top of upstream i2cdevlib that do the file renaming and changes all in one go. I was also able to cherry-pick our Encoder changes on top of upstream.

Here's my proposition for Encoder and I2Cdev. (MPU9150 is actually inside of the I2Cdev project upstream)
 - <http://github.com/pitt-ras/Encoder>
 - <http://github.com/pitt-ras/i2cdevlib> (take a look at the diffs for this, it's pretty great)

Each has an 'upstream' branch that tracks the upstream master branch, and our changes are on 'pitt-micromouse'. This puts us in a good position for dealing with future upstream changes.

Associated changes in this repo are in #85.

This was all super tedious and annoying and awful, but now I think it looks pretty good. Somebody better give me a trophy. :-)

What do you think?

amiller27, 2017-01-16:
[image]

This is the most nice-looking terrible hack I've ever seen

commit 8af65d5c (2017-01-16): Rename MPU9150 to MPU9150PittMicromouse

For our patched version.

Resolves #78.
commit a17be51f (2017-01-16): Rename I2Cdev to I2CdevPittMicromouse

For our patched version.

Prevents the conflict with upstream and the related confusion.

Resolves #78.
closed 2017-01-17 by ryanpitt