Pitt-RAS/micromouse-2016 #91: Power on hand/overheat

ryanpitt, 2017-01-17:
Randomly, and rarely. When you turn the robot on, the screen does not turn on. It's as if the program is hanging before it really has started. Meanwhile, some components on the board (underneath, front right or left) quickly overheat to skin-burning temperatures. Fortunately, it cools off quickly after powering off. Usually, the next power on is fine.

Could be an early/low-level software issue or a hardware issue.

ryanpitt, 2017-01-31:
We think overclocking to 144 MHz causes instability so that the Teensy sometimes fails to start our program.

When our program doesn't run, the range sensor outputs don't get turned off. (They default to on, unfortunately.) The way our range sensors are set up (over-voltaged), when they're left on continuously, they overheat.

If this is all true, there's probably not a good fix for this one other than awareness to turn the bot back off if the menu doesn't come up.

amiller27, 2017-01-31:
I have only noticed this happening when a program upload goes bad, have you seen it in other cases?

ryanpitt, 2017-01-31:
Yup. Every once in a while this happens to me, even if the last time the robot was on it was fine.

Figure if the processor somehow doesn't run (some sort of hardware fault??), the pins default to inputs and the range sensors stay on.

You can reproduce that much by commenting out the pin initializations in our main function and observing the overheat occur.

amiller27, 2017-01-31:
Got it, that's annoying

Riteshlogin, 2017-01-31:
You said "figure if" so I assume you aren't sure about the input pin initialization. But I can confirm - https://www.pjrc.com/teensy/td_digital.html (second paragraph)

Other than that, I assume that you guys have the range sensors on by default and the output signal turns them off somehow? But even then it seems like you guys would have to make a new PCB to really fix this and I don't think that's worth the effort for just this one thing

ryanpitt, 2017-02-02:
Agreed. It would require a PCB change, and that's not going to happen right now.