Pitt-RAS/micromouse-2016 #92: Kaos fails without moving

ryanpitt, 2017-01-17:
Seemingly random.

 1. Select KAOS
 2. Place robot
 3. Wave hand
 4. Start melody plays
 5. No movement
 6. Other melody plays (I forget which one)

ryanpitt, 2017-02-14:
@asaba96 and I haven't seen this happen since 4d2e01f2c0a6edf5a8f9ee53c64bc3919c977035. Let's say that fixed it.

ryanpitt, 2017-02-18:
Aaaand it's back folks

ryanpitt, 2017-02-28:
@asaba96 have you seen this occur since a5d17ab8bfb4d586b17b3cf150c43068473b6d84? Do we think it might be fixed again, or have you seen it?

asaba96, 2017-02-28:
So Aaron brought up a good point which is that if the second melody plays when Kaos fails, then it might be that a null path is being sent to the kaos driver or something along those lines. Basically it thinks it's done with the run and tries to move but due to it thinking it's somewhere and it not really being there the gyro heading is over 60

asaba96, 2017-02-28:
@ryanpitt so it could be anything really. I think it's worth investigating further